KMKfw/kmk_firmware #1028: [BUG] Enabling pan causes error code 10

Glodigit, 2024-09-05:
**Describe the bug**
In boot.py, I have added the following:
```
from kmk.bootcfg import bootcfg
import board
bootcfg(    sense  = board.ROW0,
            source = board.COL0,
            pan    = True,
            )
```
Unlike enabling nkro, when enabling pan, Windows 10 reports a USB descriptor error:
[image]

```
This device cannot start. (Code 10)
An unknown item was found in the report descriptor.

Problem Status:
{EXCEPTION}
Illegal Instruction
An attempt was made to execute an illegal instruction.
C000001D
```

Thus, no HID reports are processed.

**To Reproduce**
Steps to reproduce the behavior:
* Enable pan or pan + nkro
* Input: AX.P.move()

**Debug output**
If pan is enabled:
```
Traceback (most recent call last):
  File "kmk/kmk_keyboard.py", line 448, in go
  File "kmk/kmk_keyboard.py", line 483, in _init
  File "kmk/kmk_keyboard.py", line 296, in _init_hid
  File "kmk/hid.py", line 276, in __init__
  File "kmk/hid.py", line 68, in __init__
  File "kmk/hid.py", line 285, in hid_send
OSError: USB busy
```

If both nkro and pan are enabled:
```
kmk.hid: use NKRO
kmk.hid: use pan
...
kmk.keyboard: keys_pressed={MouseKey(code=1)}
kmk.hid: send: OSError: USB busy
kmk.keyboard: axes={Axis(code=3, delta=-8)}
kmk.hid: send: OSError: USB busy
```

**Additional context**
Using 
- Circuitpython 9.1.2
- Pog 1.5.2
- Windows 10

JamesToBoot, 2024-09-05:
Hry there,
Is circuitpython 9x now working? Do you have issues when you use 8x?

Glodigit, 2024-09-05:
I've never used 8x

JamesToBoot, 2024-09-06:
what to give it a try? Most MCUs have 8x available on circuitpython.org/downloads

Glodigit, 2024-09-06:
I've got a custom keyboard I reverse engineered. It seems based on a Raspberry Pi Pico. 
9.1.2 is the only version I've successfully compiled the .uf2 for, and it took multiple days to get to that point.

I'd also like to add that only enabling NKRO works.

JamesToBoot, 2024-09-06:
There are a lot of pi pico clones that are supported by CircuitPython. Attach a pic of your MCU (front and back). 

It may be that there is no compiling involved. Just a download and copy.

xs5871, 2024-09-07:
There's no indication that the CircuitPython version is to blame here and there is no reason to suggest a downgrade as the first course of action.

Glodigit, 2024-09-07:
I can only get access to the back:
[image]

- There's no SMD components on the other side.
- RP2040 as the MCU
- 2MB W25Q16JV flash chip
- LED on GP0, not GP25 like the Pico

I'll see if I can get 8.2.10 .uf2 compiled, since I've also been getting `MemoryError: memory allocation failed, allocating 1280 bytes` recently (it's always 1280 bytes) and one of the v9.0.0 notes is `New split-heap internal dynamic storage mechanism. Some CircuitPython programs may fail with MemoryError.`.

Glodigit, 2024-09-07:
8.2.x does not affect the pan bug. 
(it has fixed `MemoryError` though)
I've also tried 9.1.3 with no change.

claycooper, 2024-09-07:
Is [this](https://hackaday.io/project/197550-wk-50-trackball-keyboard) what you're building?  Are you using the files they provide or building your own?

Glodigit, 2024-09-07:
That's the project page I created where I document my findings.

PR #1069 (2025-01-08): Fix custom hid pointer report
 fixes #1028, probably?
PR #1085 (2025-02-20): Fix AC Pan descriptor
I spent an evening learning how to read USB HID descriptors and managed to fix #1028 / #1083 when noticing that AC pan is on a different page, thus seems to need additional bytes in the descriptor such as in [[1]](http://janaxelson.com/forum/index.php?topic=1302.0), [[2]](https://forum.pjrc.com/index.php?threads/teensy-3-1-mouse-horizontal-scroll.33017/).

Tested on RP2040 running CPY 9.2.4:
```
# boot.py
import board
from kmk.bootcfg import bootcfg
bootcfg(    sense  = board.BUTTON,
            midi   = False,
            pan    = True,
            nkro   = True, # False [... cut]
closed 2025-01-08 by regicidalplutophage