KMKfw/kmk_firmware #1032: [BUG] kmkfw.io No longer resolves

A-Helberg, 2024-09-30:
I'm not sure if the kmkfw.io domain is official, but it no longer resolves to any A records

**Describe the bug**
The kmkfw.io domain no longer resolves.

**To Reproduce**
`nslookup kmkfw.io`

DerArtz, 2024-10-24:
Is this going to be addressed?

xs5871, 2024-10-24:
Probably not, unless someone wants to pay for and take over maintenance for that domain. Content generation is still not automated, afaik, and it has been a bit of a pain in the past to keep up to date.

claycooper, 2024-10-24:
I want to get the site working again with automated builds but the domain and automation are in the control of other people right now and they aren't reachable.

ronaldhunt-weberedu, 2024-10-28:
The entire .io top level domain is going away. The nation it was for no longer exists, once a registration expires its over they are not allowing renewals.

mreel, 2025-02-07:
Are users supposed to just use the github repo for documentation now? I always knew it was (one of?) the most niche kb firmwares out there, but people were using it, this seems like a nail in the coffin for the whole project. 

I get it, this is a hobby for everyone and I'm not blaming any of you, but for normies like me a lack of a documentation website is the end of the road unfortunately. Good luck to everyone committed to this firmware!

ps. I know the github repo for documentation probably works just as well, but the website was barely enough as it was and, for my skill level and priorities, I need a project with better (commitment to) documentation. Thanks for all the hard work everyone is putting in, you make the kb community awesome!

claycooper, 2025-02-08:
The website was the documentation from the code repo with some stylesheets added. GitHub renders the markdown just fine and the links all work. We also keep the documentation up-to-date as the code improves.

Would I like to see the website return? Yes, but recovering the domain hasn't been successful.

mreel, 2025-02-13:
I think you missed my 'ps'. I know the github works fine, I'm not a programmer and need really good documentation to get anywhere. 
This is a me problem, not a problem I have with the kmk devs. I switched to kmk for ease of use because it's in python and I have a little python experience, but the non-extensiveness of the documentation (as compared to qmk, before the site went down), and the lack of a very active discord server made it barely any easier compared to qmk already, this is just the metaphorical straw for me personally.

claycooper, 2025-02-25:
The documentation is maintained, it's just in the firmware repo instead of on a dedicated website. We also have the Zulip server mentioned in the README as the go-to place for help instead of the chaos of a Discord server.

xgpt, 2025-02-26:
Can we not use github pages for hosting the documentation instead of just the repo file previews?