KMKfw/kmk_firmware #1033: [BUG] NKRO not working

SpoonyAllen, 2024-10-01:
N-Key Rollover does not seem to work with the latest stable KMK after enabling it in the bootcfg.py.

My setup is a simple directpin configuration with just some basic keys placed for each pin. No shift, no macro or advanced features used here. After enabling and testing with multiple input testers, i can press up to 9 buttons simultaneously but once i hit the 10th, i lose one of my pressed inputs.

xs5871, 2025-01-08:
I can not reproduce this.
The fact that you can register more than 6 keys (excluding modifiers) means NKRO is working. Can you verify that KMK is the issue and not your OS?

xs5871, 2025-02-20:
Closing because we recently reworked the HID end of KMK end NKRO is definitely working.