KMKfw/kmk_firmware #1034: Layer RGB feature is not working

parasbhanot, 2024-10-14:
i have modified the code as per the layer documentation for rgb leds as given below :

``` python
import board
import time

from kmk.kmk_keyboard import KMKKeyboard as _KMKKeyboard
from kmk.quickpin.pro_micro.sparkfun_promicro_rp2040 import pinout as pins
from kmk.scanners.keypad import KeysScanner
from kmk.extensions.media_keys import MediaKeys
from kmk.modules.macros import Macros, Tap , Delay, Press, Release
from kmk.keys import KC
from kmk.modules.layers import Layers
from kmk.extensions.rgb import RGB

_KEY_CFG_LEFT = [
     pins[9], pins[16], pins[17], pins[18], pins[19],
    pins[15], pins[14], pins[13], pins[12],  pins[0],
     pins[4],  pins[5],  pins[6],  pins[7],  pins[8],
                                  pins[10], pins[11],
]

class LayerRGB(RGB):
    def on_layer_change(self, layer):
        if layer == 0:
            self.set_hsv_fill(0, self.sat_default, self.val_default)   # red
        elif layer == 1:
            self.set_hsv_fill(170, self.sat_default, self.val_default) # blue
        elif layer == 2:
            self.set_hsv_fill(43, self.sat_default, self.val_default)  # yellow
        elif layer == 4:
            self.set_hsv_fill(0, 0, self.val_default)                  # white

class KMKKeyboard(_KMKKeyboard):
    def __init__(self):
        # create and register the scanner
        self.matrix = KeysScanner(_KEY_CFG_LEFT)

    # fmt: off
    coord_mapping = [
         0,  1,  2,  3,  4,
         5,  6,  7,  8,  9,
        10, 11, 12, 13, 14,
                    15, 16,
    ]

keyboard = KMKKeyboard()
layers = Layers()
macros = Macros()

keyboard.modules = [layers]

rgb = LayerRGB(pixel_pin=board.D1, # GPIO pin of the status LED, or background RGB light
        num_pixels=8,                # one if status LED, more if background RGB light
        rgb_order=( 1 , 0, 2),         # RGB order may differ depending on the hardware
        hue_default=0,               # in range 0-255: 0/255-red, 85-green, 170-blue
        sat_default=255,
        val_default=5,
        )
keyboard.extensions.append(rgb)

keyboard.extensions.append(MediaKeys())
keyboard.modules.append(macros)

class RGBLayers(Layers):
    def activate_layer(self, keyboard, layer, idx=None):
        super().activate_layer(keyboard, layer, idx)
        rgb.on_layer_change(layer)

    def deactivate_layer(self, keyboard, layer):
        super().deactivate_layer(keyboard, layer)
        rgb.on_layer_change(keyboard.active_layers[0])

keyboard.modules.append(RGBLayers())

# Cleaner key names
_______ = KC.TRNS
XXXXXXX = KC.NO

BaseL = KC.DF(0)
RaiseL = KC.DF(1)
RaiseL2 = KC.DF(2)
RaiseL3 = KC.DF(3)
RaiseL4 = KC.DF(4)
RaiseL5 = KC.DF(5)
RaiseL6 = KC.DF(6)
RaiseL7 = KC.DF(7)
RaiseL8 = KC.DF(8)

# Generic macros

Undo = KC.LCTL(KC.Z)
Redo = KC.LCTL(KC.Y)

# Kicad macros

GlobalLabel = KC.LCTL(KC.L)

PowerFlag = KC.MACRO(

    Tap(KC.A),
    Delay(400),
    "flag",
    Delay(100),
    Tap(KC.ENTER)

)

keyboard.keymap = [

    [  #     Base layer
           KC.Q,   KC.W,   KC.MUTE  ,    KC.VOLD,    KC.VOLU,
           RaiseL2,   KC.Q,    KC.C,   KC.R,    KC.W,
           RaiseL,  KC.Z,  KC.Z,    Undo,    Redo,
                                 KC.LGUI, KC.BSPC,
    ],

    [  #   Kicad schematics
           GlobalLabel,   KC.L,   PowerFlag,    KC.X,    KC.Y,
           KC.A,   KC.Q,    KC.C,   KC.R,    KC.W,
           BaseL,  KC.T,  KC.COMMA,    _______,    _______,
                                 KC.LGUI, KC.INS,
    ],

  ]

if __name__ == '__main__':
    keyboard.go()
```
But now colour is always red . it is not switching to blue or green as per the documentation . Am i doing something wrong ?

parasbhanot, 2024-10-14:
i have put debug statement everywhere and go the following serial logs 

``` bash 
Code stopped by auto-reload. Reloading soon.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
2527502 kmk.keyboard: Initialising KMKKeyboard
2527505 kmk.hid: use 6KRO
2527506 kmk.hid: use no pan
2527507 kmk.keyboard: hid=USBHID(REPORT_BYTES=9)
2527510 kmk.keyboard: matrix=['KeysScanner']
2527513 kmk.keyboard: modules=['Macros', 'RGBLayers']
2527548 kmk.extensions.rgb: pixels[0] = <class 'NeoPixel'>[8]
2527550 kmk.keyboard: extensions=['LayerRGB', 'MediaKeys']
2527575 kmk.keyboard: mem_info used:76800 free:95616
2528704 kmk.keyboard: <Event: key_number 10 pressed>: LayerKey
2528707 kmk.keyboard: coordkeys_pressed={10: LayerKey}
Activating layer: 1
2528710 kmk.modules.layers: active_layers=[1]
Layer change callback: 1
Setting color to blue
2528899 kmk.keyboard: <Event: key_number 10 released>: LayerKey
2528901 kmk.keyboard: coordkeys_pressed={}
2555139 kmk.keyboard: cleaning up...
2555144 kmk.keyboard: ...done
```

and my updated code is 

```
import board

import time

from kmk.kmk_keyboard import KMKKeyboard as _KMKKeyboard
from kmk.quickpin.pro_micro.sparkfun_promicro_rp2040 import pinout as pins
from kmk.scanners.keypad import KeysScanner
from kmk.extensions.media_keys import MediaKeys
from kmk.modules.macros import Macros, Tap , Delay, Press, Release
from kmk.keys import KC
from kmk.extensions.rgb import RGB
from kmk.modules.layers import Layers

_KEY_CFG_LEFT = [
     pins[9], pins[16], pins[17], pins[18], pins[19],
    pins[15], pins[14], pins[13], pins[12],  pins[0],
     pins[4],  pins[5],  pins[6],  pins[7],  pins[8],
                                  pins[10], pins[11],
]

class LayerRGB(RGB):
    def on_layer_change(self, layer):
        print(f"Layer change callback: {layer}")  # Debug statement
        if layer == 0:
            print("Setting color to red")
            self.set_hsv_fill(0, self.sat_default, self.val_default) # red

        elif layer == 1:
            print("Setting color to blue")
            self.set_hsv_fill(170, self.sat_default, self.val_default)  # blue

        elif layer == 2:
            print("Setting color to orange")
            self.set_hsv_fill(43, self.sat_default, self.val_default)  # yellow
        elif layer == 4:
            print("Setting color to yellow")
            self.set_hsv_fill(0,0 , self.val_default)  # white

class KMKKeyboard(_KMKKeyboard):
    def __init__(self):
        # create and register the scanner
        self.matrix = KeysScanner(_KEY_CFG_LEFT)

    # fmt: off
    coord_mapping = [
         0,  1,  2,  3,  4,   
         5,  6,  7,  8,  9,   
        10, 11, 12, 13, 14,
                    15, 16,
    ]

keyboard = KMKKeyboard()
# layers = Layers()
macros = Macros()

rgb = LayerRGB(pixel_pin=board.D1, # GPIO pin of the status LED, or background RGB light
        num_pixels=8,                # one if status LED, more if background RGB light
        rgb_order=( 1 , 0, 2),         # RGB order may differ depending on the hardware
        hue_default=0,               # in range 0-255: 0/255-red, 85-green, 170-blue
        sat_default=255,
        val_default=5,
        )

class RGBLayers(Layers):
    def activate_layer(self, keyboard, layer, idx=None):
        print(f"Activating layer: {layer}")  # Debug statement
        super().activate_layer(keyboard, layer, idx)
        rgb.on_layer_change(layer)

    def deactivate_layer(self, keyboard, layer):
        print(f"Deactivating layer: {layer}")  # Debug statement
        super().deactivate_layer(keyboard, layer)
        rgb.on_layer_change(keyboard.active_layers[0])

keyboard.extensions.append(rgb)
keyboard.extensions.append(MediaKeys())
keyboard.modules.append(macros)
# keyboard.modules.append(layers)
keyboard.modules.append(RGBLayers())

# Cleaner key names
_______ = KC.TRNS
XXXXXXX = KC.NO

BaseL = KC.DF(0) 
RaiseL1 = KC.DF(1)  
RaiseL2 = KC.DF(2)  
RaiseL3 = KC.DF(3)  
RaiseL4 = KC.DF(4)  
RaiseL5 = KC.DF(5)  
RaiseL6 = KC.DF(6)  
RaiseL7 = KC.DF(7)  
RaiseL8 = KC.DF(8)  

# Generic macros 

Undo = KC.LCTL(KC.Z)
Redo = KC.LCTL(KC.Y)

# Kicad macros 

GlobalLabel = KC.LCTL(KC.L)

PowerFlag = KC.MACRO(
    Tap(KC.A),
    Delay(400),
    "flag",
    Delay(100),
    Tap(KC.ENTER)
)

keyboard.keymap = [

    [  #     Base layer
           KC.Q,   KC.W,   KC.MUTE	,    KC.VOLD,    KC.VOLU,
           KC.N,   KC.Q,    KC.C,   KC.R,    KC.W,
           RaiseL1,  KC.Z,  KC.Z,    Undo,    Redo,    
                                 KC.LGUI, KC.BSPC,
    ],

    [  #   Kicad schematics
           GlobalLabel,   KC.L,   PowerFlag,    KC.X,    KC.Y,
           KC.A,   KC.Q,    KC.C,   KC.R,    KC.W,
           BaseL,  KC.T,  KC.COMMA,    _______,    _______,    
                                 KC.LGUI, KC.INS,
    ],

  ]

if __name__ == '__main__':
    keyboard.go()
```

xs5871, 2024-10-14:
If you don't have an active animation you have to apply the color change manually by calling the rgb extension's `show()` method.

PR #1035 (2024-10-14): Update layers.md
resolves #1034
closed 2024-10-14 by claycooper