KMKfw/kmk_firmware #1039: [BUG] Hold-Tap doesn't appear to work with transparent keycodes

hariganti, 2024-10-30:
**Describe the bug**
When using hold-tap (`KC.HT(KC.TAP, KC.HOLD)`), using `KC.TRNS` for the tap action doesn't properly send the keycode that should be sent.

**To Reproduce**
Steps to reproduce the behavior:
```
from kmk.keys import KC
CUST = KC.HT(KC.TRNS, KC.LCTL)
Keyboard.keymap = [
  [ ... KC.A, KC.B ...],
  [ ... CUST, CUST ...]
]
```

1. Activate second layer
2. Press 'B'
3. Press 'A'
4. Release 'A'
5. Release 'B'

**Expected behavior**
Control + A (select all) should be the result

**Debug output**
No output available at this time

**Additional context**
What I'd like to do is have a layer with home row mods, so holding a key will momentarily enable that layer, then holding one of the hold-tap keys will give me the desired modifier. As I have other layers, using the transparency keycode for the tap function is desirable since it will allow me to use the home row mods on whichever layer I was actively using before activating the momentary layer. This is also why I can't just do something like, `KC.HT(KC.A, KT.LCTL)`, since that won't allow me to use control as a modifier on all layers

xs5871, 2024-10-31:
That's a feature request. The transparent key resolution happens at a completely different part of the code.

hariganti, 2024-10-31:
> Thanks for the correction. It appeared like a bug since I didn't see anything to indicate that the transparency keycode couldn't be used there in the way I tried