KMKfw/kmk_firmware #1040: [BUG] LM crashes when used as the emitted key from a chord

hariganti, 2024-10-30:
**Describe the bug**
When using LM as the emitted key from a chord, KMK appears to crash and restart

**To Reproduce**
```
from kmk.keys import KC
from kmk.modules.layers import Layers
from kmk.modules.combos import Combos, Chord

combos = Combos()
combos.combos = [
  Chord((1, 2), KC.LM(1, KC.LCTL), match_coord = True, timeout = 1000)
]

keyboard.keymap = [
  [KC.A, KC.B, KC.C ... ],
  [KC.S, KC.T, KC.P ...] 
]
```

1. Press 'A'
2. Press 'B'
3. Press 'C' (3rd key on base layer)

Following the above, KMK appears to crash and restarts

**Expected behavior**
Control + P (print) should be the result

**Debug output**
Unable to connect to serial terminal for debug output

**Additional context**
What I'd like to do is have a layer with home row mods, so a combination of a leader key and the appropriate home row mod will enable the home row mod but on a different layer so I can use a non-QWERTY layout but keep keybindings common. For example, if I use the ISRT layout, then Control + Z (on a QWERTY layout) would end up entered as Control + Q, so I want to switch to a QWERTY layout when using mods. This is partly because I can't seem to find a way to get something like the following working:
```
combos.combos = [
  Chord((1, 2), KC.LCTL, match_coord = True, timeout = 1000),
  Chord((KC.LCTL, KC.Q), KC.LCTL(KC.Z), timeout = 1000)
]
```

xs5871, 2024-10-31:
If something crashes it's a good idea to also post the debug output.

hariganti, 2024-10-31:
I understand, but as I put in the initial information, I've been unable to connect to the serial monitor so far. For some reason, using Mu didn't work even though I am part of the dialout group, and I haven't had time to work out another means of connecting

hariganti, 2024-11-01:
Ok, so I got `tio` working and I was able to display the serial output, but now, the error seems to have changed. The initial error of it crashing during operation has been replaced by crashing immediately after attempting to start. I get the following error:

```
code.py output:
Starting
210170 kmk.keys: Invalid key: LM
Traceback (most recent call last):
  File "code.py", line 78, in <module>
TypeError: 'Key' object is not callable
```

Interestingly, even when I remove the `Chord` from my combos, I get a new error when connected to the serial monitor:

```
code.py output:
Starting
246324 kmk.keyboard: Initialising KMKKeyboard
246327 kmk.hid: use 6KRO
246335 kmk.hid: use no pan
246336 kmk.keyboard: hid=USBHID(REPORT_BYTES=9)
246339 kmk.keyboard: matrix=['MatrixScanner']
246344 kmk.keyboard: modules=['Combos', 'HoldTap', 'LEDLayers', 'StickyKeys']
Traceback (most recent call last):
  File "kmk/kmk_keyboard.py", line 345, in during_bootup
AttributeError: 'DigitalInOut' object has no attribute 'during_bootup'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "kmk/kmk_keyboard.py", line 448, in go
  File "kmk/kmk_keyboard.py", line 486, in _init
  File "kmk/kmk_keyboard.py", line 347, in during_bootup
  File "kmk/kmk_keyboard.py", line 26, in debug_error
AttributeError: 'DigitalInOut' object has no attribute '__module__'
246351 kmk.keyboard: cleaning up...
Traceback (most recent call last):
  File "kmk/kmk_keyboard.py", line 442, in deinit
TypeError: function takes 1 positional arguments but 2 were given

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "code.py", line 229, in <module>
  File "kmk/kmk_keyboard.py", line 462, in go
  File "kmk/kmk_keyboard.py", line 444, in deinit
  File "kmk/kmk_keyboard.py", line 26, in debug_error
AttributeError: 'DigitalInOut' object has no attribute '__module__'

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

Also, KMK seems to take much longer to start up now. It used to start up almost immediately on a reset, but now it takes almost 30 seconds before the keyboard sends keystrokes to the OS.

hariganti, 2025-02-20:
I don't recall. I should just cancel this issue since I was using KMK for some early prototyping, but I'm not using this code anymore