KMKfw/kmk_firmware #1041: [Enhancement] Title

dowling-john, 2024-11-01:
**Is your feature request related to a problem? Please describe.**
It would be good to be able to split keys from their shift pairs

**Describe the solution you'd like**
For instance 9 and ( should be assignable to seperate keys

xs5871, 2024-11-01:
You can already do that, so I'm assuming you actually want #409.

dowling-john, 2024-11-02:
Thanks for the reply.

I am currently using a macro to define a single key press to get the ( symbol from a single key press.

```
LBRACK = MACRO(
   Press(KC.LSHFT)
   Tap(KC.N9) 
   Release(KC.LSHFT)
)
```

Is this the right way to tackle this problem or is there something that I have missed ?

xs5871, 2024-11-02:
https://github.com/KMKfw/kmk_firmware/blob/main/docs/en/keycodes.md#us-ansi-shifted-symbols

dowling-john, 2024-11-02:
How in the heck did I miss that 🤯 

Thanks

closed 2024-11-02 by dowling-john