KMKfw/kmk_firmware #1062: [BUG]  Analog module does not work

doman18, 2024-12-22:
My goal is to create gaming keypad with joystick and 26 keys similar to Razer Tartarus, Hori TAC, Azeron Cyborg, Saitek PZ31a etc. Im using Pi Pico (for prototyping) and later will use Pi Zero. 

Im following this
https://github.com/LennartEd/Analog-WASD-Keyboard-kmk-pi-pico-rp2040/blob/main/README.md

I want to use my joystick the same way author did - by assigning WASD to it. 
I adjusted my code so it looks like this:
```
import board

from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.scanners import DiodeOrientation
from kmk.modules.layers import Layers
from kmk.modules.analog import AnalogKey #handles analoginput

keyboard = KMKKeyboard()
layers = Layers()
ana = AnalogKey()

keyboard.modules = [layers,ana]

keyboard.col_pins = (board.GP18,board.GP19)
keyboard.row_pins = (board.GP12,board.GP13)
keyboard.diode_orientation = DiodeOrientation.COL2ROW

keyboard.keymap = [
    [
        KC.A, KC.B,
        KC.C, KC.D,
    ]
]

#ANALOG 
ana.pins = (board.A0,board.A1) #define analog pins

ana.reverse = False #flips direction(standard is: analog value goes down when key pressed) 

# define per layer if keypresses should be Threshhold(0) or Rapid trigger(1) explanation: https://wooting.io/rapid-trigger 
ana.keypType = (1,1)

#threshold: when analog value is passed key is pressed 
#rapit: when a certain distance is surpaced key is pressed (the lower the value the faster responce but at a certain point noise can trigger keypress)
    #use the provided "maxChangeInVal.py"+ 300 to find a good rapid trigger value
ana.threshold = (30000,1800) #default (30000,1800) 

ana.map = [
    [
        KC.W,KC.A,KC.S,KC.D
    ]
]

if __name__ == '__main__':
    keyboard.go()
```

Unfortunately im geting an error:
```
code.py output:
Starting
Traceback (most recent call last):
  File "code.py", line 9, in <module>
ImportError: no module named 'kmk.modules.analog'
```

Is there other way to achieve what i want?

xs5871, 2024-12-22:
This isn't a bug. The *draft* PR that implements that module isn't merged yet.

piman13, 2024-12-30:
the project you are referencing anyway made their own module that seem to be very application specific. and while it is interesting it is not likely something that would be merged into kmk for everyone.
in the mean time though you could use the one provided by the project you are referencing

piman13, 2024-12-30:
long term however the is work on analogio and gamepad HID so kmk being used for this application would definitely be a possibility in the future.

(I kinda think a hotas controller would be cool now.... one project at a time....)

doman18, 2025-01-07:
@piman13 AFAIK gamepad HID does not support matrixes. At least i havent found info about it. And in my case (26 keys) matrix is mandatory.

That the main problem here - to get some matrix library with joystick support.

After some research and help on adafruit forums i came out with this code (based on analogio and adafruit_hid library):

```
import board, keypad, usb_hid, time, analogio
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode

##### JOYSTICK

# Connect an analog two-axis joystick to A0 and A1.
# https://how2electronics.com/how-to-use-adc-in-raspberry-pi-pico-adc-example-code/
ax = analogio.AnalogIn(board.A0)
ay = analogio.AnalogIn(board.A1)

# Equivalent of Arduino's map() function.
def range_map(x, in_min, in_max, out_min, out_max):
    return (x - in_min) * (out_max - out_min) // (in_max - in_min) + out_min

deadzoneSize=15

##### KEYBOARD
kbd = Keyboard(usb_hid.devices)

km = keypad.KeyMatrix(
    row_pins=(board.GP18,board.GP19),
    column_pins=(board.GP12,board.GP13),
)

mapping_list = [Keycode.W,Keycode.A,Keycode.S,Keycode.D]

xKey="none"
yKey="none"

while True:
    x=range_map(ax.value, 0, 65535, -127, 127)
    y=range_map(ay.value, 0, 65535, -127, 127)
 

    if x>deadzoneSize and xKey != "D": #
        kbd.press(Keycode.D)
        xKey="D"
    elif x<-deadzoneSize and xKey != "A":
        kbd.press(Keycode.A)
        xKey="A"
    elif abs(x)<=deadzoneSize and xKey != "none":
        xKey="none"
        kbd.release(Keycode.A)
        kbd.release(Keycode.D)

    if y>deadzoneSize and yKey != "W":
        kbd.press(Keycode.W)
        yKey="W"
    elif y<-deadzoneSize and yKey != "S":
        kbd.press(Keycode.S)
        yKey="S"
    elif abs(y)<=deadzoneSize and yKey != "none":
        yKey="none"
        kbd.release(Keycode.W)
        kbd.release(Keycode.S)

    print(" x", xKey, "y", yKey)

    event = km.events.get()
    if event:
        if event.pressed:
            kbd.press(mapping_list[event.key_number])
        else:
            kbd.release(mapping_list[event.key_number])
```

https://forums.adafruit.com/viewtopic.php?p=1040009#p1040009

earvingad, 2025-02-07:
@doman18 does your code work inside of the `code.py`?

doman18, 2025-02-10:
@earvingad Yes, i copy/pasted it from my code.py file on my RP Pico. I later tried it on my RP Zero (after adjusting some pins) and it worked as well. 

Still did not use it in full solution as had to drop the "handwiring" idea, learn KiCAD, re-create my design there and order my first pcb. Currently im waiting for delivery and once i get it i will continue to work on this code.

xs5871, 2025-02-20:
Closing because analog module has been merged.

closed 2025-02-20 by xs5871