KMKfw/kmk_firmware #1071: [BUG] Layer sometimes gets stuck

domiSchenk, 2025-01-10:
**Describe the bug**
Sometimes when i change layer with KC.LT the layer does not switch back to the default layer

**To Reproduce**
I'm still not sure why this happens and how to reproduce it consistently.
In this example I wanted to activate KC.CW, so I switched to layer [4] to press KC.CW.
To do this, I press key 18 (left middle thumb) and then key 29 (right, middle left) and released key 18.
After that I was stuck in layer [4, 0].
See output: 
there was no `kmk.modules.layers: active_layers=[0]` after `185591 kmk.modules.layers: active_layers=[4, 0]`

**Expected behavior**
With LT key is released it should return to default layer.

**Debug output**
```
173118 kmk.keyboard: keys_pressed={KeyboardKey(code=8)}
175081 kmk.keyboard: <Event: key_number 18 pressed>: HoldTapKey
175084 kmk.keyboard: coordkeys_pressed={18: HoldTapKey}
175197 kmk.keyboard: <Event: key_number 18 released>: HoldTapKey
175200 kmk.keyboard: coordkeys_pressed={}
175202 kmk.modules.holdtap: ht_activate_tap
175203 kmk.modules.holdtap: ht_deactivate_tap
175207 kmk.keyboard: keys_pressed={KeyboardKey(code=44)}
178833 kmk.keyboard: <Event: key_number 18 pressed>: HoldTapKey
178836 kmk.keyboard: coordkeys_pressed={18: HoldTapKey}
179071 kmk.keyboard: <Event: key_number 29 pressed>: KeyboardKey(code=16)
179075 kmk.modules.holdtap: ht_activate_on_interrupt
179076 kmk.modules.holdtap: ht_activate_hold
179077 kmk.keyboard: coordkeys_pressed={18: HoldTapKey, 29: None}
179083 kmk.modules.layers: active_layers=[4, 0]
179086 kmk.keyboard: coordkeys_pressed={18: HoldTapKey, 29: TapDanceKey}
179209 kmk.keyboard: <Event: key_number 29 released>: TapDanceKey
179212 kmk.keyboard: coordkeys_pressed={18: HoldTapKey}
179290 kmk.modules.holdtap: ht_activate_tap
179291 kmk.modules.holdtap: ht_deactivate_tap
179299 kmk.keyboard: <Event: key_number 18 released>: HoldTapKey
179302 kmk.keyboard: coordkeys_pressed={}
179304 kmk.modules.holdtap: ht_deactivate_on_interrupt
179304 kmk.modules.holdtap: ht_deactivate_hold
179310 kmk.modules.layers: active_layers=[0]
179755 kmk.keyboard: <Event: key_number 23 pressed>: KeyboardKey(code=15)
179759 kmk.keyboard: coordkeys_pressed={23: KeyboardKey(code=15)}
179763 kmk.keyboard: keys_pressed={ModifierKey(code=2), KeyboardKey(code=15)}
179853 kmk.keyboard: <Event: key_number 23 released>: KeyboardKey(code=15)
179858 kmk.keyboard: coordkeys_pressed={}
180081 kmk.keyboard: <Event: key_number 5 pressed>: TapDanceKey
180085 kmk.keyboard: coordkeys_pressed={5: TapDanceKey}
180217 kmk.keyboard: <Event: key_number 5 released>: TapDanceKey
180220 kmk.keyboard: coordkeys_pressed={}
180288 kmk.modules.holdtap: ht_activate_tap
180290 kmk.modules.holdtap: ht_deactivate_tap
180294 kmk.keyboard: keys_pressed={ModifierKey(code=2), KeyboardKey(code=4)}
181350 kmk.keyboard: <Event: key_number 1 pressed>: KeyboardKey(code=26)
181356 kmk.keyboard: coordkeys_pressed={1: KeyboardKey(code=26)}
181359 kmk.keyboard: keys_pressed={ModifierKey(code=2), KeyboardKey(code=26)}
181528 kmk.keyboard: <Event: key_number 1 released>: KeyboardKey(code=26)
181532 kmk.keyboard: coordkeys_pressed={}
181807 kmk.keyboard: <Event: key_number 26 pressed>: HoldTapKey
181811 kmk.keyboard: coordkeys_pressed={26: HoldTapKey}
181904 kmk.keyboard: <Event: key_number 26 released>: HoldTapKey
181907 kmk.keyboard: coordkeys_pressed={}
181909 kmk.modules.holdtap: ht_activate_tap
181910 kmk.modules.holdtap: ht_deactivate_tap
181916 kmk.keyboard: keys_pressed={ModifierKey(code=2), KeyboardKey(code=12)}
183381 kmk.keyboard: <Event: key_number 7 pressed>: HoldTapKey
183385 kmk.keyboard: coordkeys_pressed={7: HoldTapKey}
183518 kmk.keyboard: <Event: key_number 7 released>: HoldTapKey
183521 kmk.keyboard: coordkeys_pressed={}
183523 kmk.modules.holdtap: ht_activate_tap
183524 kmk.modules.holdtap: ht_deactivate_tap
183530 kmk.keyboard: keys_pressed={ModifierKey(code=2), KeyboardKey(code=22)}
185297 kmk.keyboard: <Event: key_number 18 pressed>: HoldTapKey
185301 kmk.keyboard: coordkeys_pressed={18: HoldTapKey}
185579 kmk.keyboard: <Event: key_number 29 pressed>: KeyboardKey(code=16)
185583 kmk.modules.holdtap: ht_activate_on_interrupt
185584 kmk.modules.holdtap: ht_activate_hold
185586 kmk.keyboard: coordkeys_pressed={18: HoldTapKey, 29: None}
185591 kmk.modules.layers: active_layers=[4, 0]
185595 kmk.keyboard: coordkeys_pressed={18: HoldTapKey, 29: TapDanceKey}
185657 kmk.keyboard: <Event: key_number 29 released>: TapDanceKey
185661 kmk.keyboard: coordkeys_pressed={18: HoldTapKey}
185746 kmk.keyboard: <Event: key_number 18 released>: HoldTapKey
185749 kmk.modules.holdtap: ht_activate_tap
185750 kmk.modules.holdtap: ht_deactivate_tap
185752 kmk.keyboard: coordkeys_pressed={}
185757 kmk.keyboard: release w/o press:18
185758 kmk.keyboard: coordkeys_pressed={}
186672 kmk.keyboard: <Event: key_number 29 pressed>: TapDanceKey
186676 kmk.keyboard: coordkeys_pressed={29: TapDanceKey}
186832 kmk.keyboard: <Event: key_number 29 released>: TapDanceKey
186835 kmk.keyboard: coordkeys_pressed={}
186879 kmk.modules.holdtap: ht_activate_tap
186880 kmk.modules.holdtap: ht_deactivate_tap
187578 kmk.keyboard: <Event: key_number 29 pressed>: TapDanceKey
187582 kmk.keyboard: coordkeys_pressed={29: TapDanceKey}
187806 kmk.modules.holdtap: ht_activate_hold
187813 kmk.keyboard: <Event: key_number 29 released>: TapDanceKey
187816 kmk.keyboard: coordkeys_pressed={}
187817 kmk.modules.holdtap: ht_deactivate_hold
188203 kmk.keyboard: <Event: key_number 29 pressed>: TapDanceKey
188206 kmk.keyboard: coordkeys_pressed={29: TapDanceKey}
188318 kmk.keyboard: <Event: key_number 29 released>: TapDanceKey
188321 kmk.keyboard: coordkeys_pressed={}
188409 kmk.modules.holdtap: ht_activate_tap
188432 kmk.modules.holdtap: ht_deactivate_tap
188475 kmk.keyboard: <Event: key_number 29 pressed>: TapDanceKey
188479 kmk.keyboard: coordkeys_pressed={29: TapDanceKey}
188572 kmk.keyboard: <Event: key_number 29 released>: TapDanceKey
188575 kmk.keyboard: coordkeys_pressed={}
188682 kmk.modules.holdtap: ht_activate_tap
188684 kmk.modules.holdtap: ht_deactivate_tap
188691 kmk.keyboard: <Event: key_number 29 pressed>: TapDanceKey
188715 kmk.keyboard: coordkeys_pressed={29: TapDanceKey}
188806 kmk.keyboard: <Event: key_number 29 released>: TapDanceKey
188809 kmk.keyboard: coordkeys_pressed={}
188918 kmk.modules.holdtap: ht_activate_tap
188919 kmk.modules.holdtap: ht_deactivate_tap
189200 kmk.keyboard: <Event: key_number 29 pressed>: TapDanceKey
189204 kmk.keyboard: coordkeys_pressed={29: TapDanceKey}
189277 kmk.keyboard: <Event: key_number 29 released>: TapDanceKey
189281 kmk.keyboard: coordkeys_pressed={}
189376 kmk.keyboard: <Event: key_number 29 pressed>: TapDanceKey
189380 kmk.keyboard: coordkeys_pressed={29: TapDanceKey}
189472 kmk.keyboard: <Event: key_number 29 released>: TapDanceKey
189475 kmk.keyboard: coordkeys_pressed={}
189582 kmk.modules.holdtap: ht_activate_tap
189583 kmk.modules.holdtap: ht_deactivate_tap
189588 kmk.keyboard: keys_pressed={KeyboardKey(code=57)}
189599 kmk.keyboard: <Event: key_number 29 pressed>: TapDanceKey
189602 kmk.keyboard: coordkeys_pressed={29: TapDanceKey}
189688 kmk.keyboard: <Event: key_number 29 released>: TapDanceKey
189691 kmk.keyboard: coordkeys_pressed={}
189805 kmk.modules.holdtap: ht_activate_tap
189807 kmk.modules.holdtap: ht_deactivate_tap
190841 kmk.keyboard: <Event: key_number 29 pressed>: TapDanceKey
190845 kmk.keyboard: coordkeys_pressed={29: TapDanceKey}
191048 kmk.modules.holdtap: ht_activate_hold
191067 kmk.keyboard: <Event: key_number 29 released>: TapDanceKey
191070 kmk.keyboard: coordkeys_pressed={}
191072 kmk.modules.holdtap: ht_deactivate_hold
193068 kmk.keyboard: <Event: key_number 29 pressed>: TapDanceKey
193071 kmk.keyboard: coordkeys_pressed={29: TapDanceKey}
193204 kmk.keyboard: <Event: key_number 29 released>: TapDanceKey
193207 kmk.keyboard: coordkeys_pressed={}
193273 kmk.modules.holdtap: ht_activate_tap
193275 kmk.modules.holdtap: ht_deactivate_tap
194736 kmk.keyboard: <Event: key_number 5 pressed>: ModifierKey(code=8)
194741 kmk.keyboard: coordkeys_pressed={5: ModifierKey(code=8)}
194744 kmk.keyboard: keys_pressed={ModifierKey(code=8)}
194930 kmk.keyboard: <Event: key_number 5 released>: ModifierKey(code=8)
194935 kmk.keyboard: coordkeys_pressed={}

```

**Additional context**
Repo to my config is here: https://github.com/domiSchenk/DOS-Keyboard

domiSchenk, 2025-01-17:
to reproduce:
- momentarily change layer 
- click a key with a TD definition
- release LT key fast, nearly simultaneously

Debug message  release w/o press:XX appears. Layer is not switched back to 0

 following config does the same with the same problem:
```
KC.HT(KC.ENT, KC.MO(8), prefer_hold=False, tap_interrupted=True, tap_time=150)
```

xs5871, 2025-01-19:
I'm in the process of looking into it. It's a bit convoluted, the current fix I have breaks combos.

PR #1076 (2025-02-03): Fix inconsistencies with TD on momentary layers
resolves #1071
closed 2025-02-03 by regicidalplutophage