KMKfw/kmk_firmware #1075: [BUG] Examples in "keys" documentation crash keyboard with attribute error

robotfishe, 2025-01-31:
**Describe the bug**
When a key assigned a custom keycode is pressed, the keyboard crashes with the following message in serial output:
`kmk.kmk_keyboard: Unexpected error: AttributeError: 'LimitKey' object has no attribute 'code'`

**To Reproduce**
Copy and paste the following example code from keys.md to main.py:
```
from kmk.keys import Key

class LimitKey(Key):
    def __init__(self, key, limit):
        self.key = KC.A
        self.limit = limit

    def on_press(self, keyboard, coord_int=None):
        if self.limit > 0:
            keyboard.add_key(self.key)

    def on_release(self, keyboard, coord_int=None):
        if self.limit > 0:
            self.limit -= 1
            keyboard.remove_key(self.key)

KC_A10 = LimitKey(KC.A, 10)
KC_B20 = LimitKey(KC.B, 20)

```
Assign the code KC_A10 to spot in the keymap. Press the key.

**Expected behavior**
According to the example documentation, this should output an "a" the first ten times it's pressed, then not respond to subsequent presses.

**Debug output**
See above.

**Additional context**
I found another example online at https://www.reddit.com/r/olkb/comments/1gcvg4e/shift_lock_in_kmk/ which seems to have once worked but now also throws the same error. Documentation might need to be updated for a change in how KMK handles key input?

robotfishe, 2025-02-01:
After some further experimentation I was able to get the example working with the following changes (additions in bold):

```
class LimitKey(Key):
    def __init__(self, key, limit):
        self.key = key
        self.limit = limit
        **self.code = 50
        self.has_modifiers=0**

    def on_press(self, keyboard, coord_int=None):
        if self.limit > 0:
            keyboard.add_key(self.key)

    def on_release(self, keyboard, coord_int=None):
        if self.limit > 0:
            self.limit -= 1
            keyboard.remove_key(self.key)

KC_A10 = LimitKey(KC.A, 10)
KC_B20 = LimitKey(KC.B, 20)
```

xs5871, 2025-02-01:
Have you tried using a current version of KMK?

robotfishe, 2025-02-01:
> Have you tried using a current version of KMK?

It's a new install using files downloaded from Github on 29 January, so unless there's been a big change to the Key class in the last few days, that shouldn't be the problem.

xs5871, 2025-02-01:
Your copy is at least 6 months old.

robotfishe, 2025-02-01:
> Your copy is at least 6 months old.

I see it now - looks like I downloaded the "master" branch instead of "main". I must have followed an old link from somewhere since it looks like main took over 10 months ago. Anyway, with the current main branch installed, the custom key example is working without the extra attributes set. I'll close this issue, but can I suggest Getting_Started.md in the master branch be edited with a link to main? The existence of both those branch names has a lot of potential to confuse.

closed 2025-02-01 by robotfishe