KMKfw/kmk_firmware #1084: [BUG] KC.APP key doesn't work

BillyBuerger, 2025-02-16:
**Describe the bug**
I'm trying to implement the Windows Menu/App key in my keyboard layout.  I don't see it mentioned in the keys.md but I see it listed in a number of boards in the repo.  I used KC.APP as these others have (and as QMK uses) but it doesn't do anything.  I did some searching through the keys.py file and don't see it there either.  Is it just not implemented?

**To Reproduce**
Steps to reproduce the behavior:
* Include KC.APP in your keymap
* press key

**Expected behavior**
The Windown Menu key code to be sent

**Debug output**
NA

**Additional context**
NA

xs5871, 2025-02-18:
It's undocumented and in `kmk.extensions.international`.

BillyBuerger, 2025-02-18:
Thanks, that was it.

closed 2025-02-18 by BillyBuerger